ethz-asl/kalibr #585: error: Spline Coefficient Buffer Exceeded. Set larger buffer margins! with static_dataset

zoldaten, 2022-12-07:
hi!
i`m trying to calibrate static dataset with rosbag recorded on raspberry pi:
```
rosrun kalibr kalibr_calibrate_rs_cameras --bag chess_static_30fps.bag --model pinhole-radtan-rs --target checkerboard_8x6.yaml \
--topic /camera/image_raw --inverse-feature-variance 1 --frame-rate 30
```

and got this:
```
Exception in thread block: [aslam::Exception] /home/ubuntu/kalibr_workspace/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [1.6704e+09 <= 1.6704e+09 < 1.6704e+09]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
Exception in thread block: [aslam::Exception] /home/ubuntu/kalibr_workspace/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [1.6704e+09 <= 1.6704e+09 < 1.6704e+09]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
Exception in thread block: [aslam::Exception] /home/ubuntu/kalibr_workspace/src/kalibr/aslam_nonparametric_estimation/aslam_splines/src/BSplineExpressions.cpp:447: toTransformationMatrixImplementation() assert(_bufferTmin <= _time.toScalar() < _bufferTmax) failed [1.6704e+09 <= 1.6704e+09 < 1.6704e+09]: Spline Coefficient Buffer Exceeded. Set larger buffer margins!
[ERROR] [1670403011.520368]: Exception: std::exception
```

The camera i use is csi fisheye 160 deg. And it seems i got enough good pictures to calibrate:
```
Number of images: 3596
Extracting calibration target corners
  Extracted corners for 339 images (of 3596 images)
```

what`s going wrong ?

goldbattle, 2022-12-07:
Please pose a bag and configuration.

zoldaten, 2022-12-07:
checkerboard_8x6.yaml :
```
#checkerboard_8x6.yaml
target_type: 'checkerboard'
targetCols: 8
targetRows: 6
rowSpacingMeters: 0.027
colSpacingMeters: 0.027
```

rosbag - [link](https://drive.google.com/file/d/1I_j-zvaok1qSMts9eahB4SKabUeqppQN/view?usp=sharing)

command i use:
`rosrun kalibr kalibr_calibrate_rs_cameras --bag chess_static_30fps.bag --model pinhole-radtan-rs --target checkerboard_8x6.yaml --topic /camera/image_raw --inverse-feature-variance 1 --frame-rate 30`