ethz-asl/kalibr #586: difference Kalibr with classic ROS wiki mono camera calibration

zoldaten, 2022-12-07:
hi!
i have calibrated camera fisheye 160 degree with ROS wiki tutorial https://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration.
Chessboard (checkerboard in Kalibr terms) used.
Got this result:
```
---ost.yaml---
image_width: 320
image_height: 240
camera_name: head_camera
camera_matrix:
  rows: 3
  cols: 3
  data: [118.64993,  -0.0212 , 160.43422,
           0.     , 120.44204, 119.19462,
           0.     ,   0.     ,   1.     ]
distortion_model: equidistant
distortion_coefficients:
  rows: 1
  cols: 4
  data: [-0.080996, -0.082818, 0.730590, -0.286382]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1., 0., 0.,
         0., 1., 0.,
         0., 0., 1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [118.64993,  -0.0212 , 160.43422,   0.     ,
           0.     , 120.44204, 119.19462,   0.     ,
           0.     ,   0.     ,   1.     ,   0.     ]
```

Then i use Kalibr tutorial, record rosbag for the same camera and calibrated it:
```
rosrun kalibr kalibr_calibrate_rs_cameras --bag chess_static_30fps.bag --model pinhole-radtan-rs --target checkerboard_8x6.yaml \
    --topic /usb_cam/image_raw --inverse-feature-variance 1 --frame-rate 30
```
As i dont know exectly which model of rs camera i use i did calibration for all models also.
And got results:
```
---with apriltag---
pinhole-radtan-rs
Intrinsics:
[155.30652299 154.95205584 151.209779   127.89140297]
Distortion:
[-0.2860585  0.0596586  0.000921  -0.0014569]

pinhole-equi-rs
LineDelay:
9.534137589259335e-05
Intrinsics:
[155.19988577 154.81930505 149.59845224 130.08993271]
Distortion:
[-0.02752885 -0.01670695  0.01461811 -0.00684478]

omni-radtan-rs
LineDelay:
0.0001388888888888889
Intrinsics:
[  1.        256.9815044 256.9815044 159.5       119.5      ]
Distortion:
[0. 0. 0. 0.]
```

*I tried to use chessboard but crashed with error Spline Coefficient Buffer Exceeded. Set larger buffer margins!

So the question is:
why the difference with ROS wiki presents ? 

*i`ve tested calibration with orb_slam3. ROS wiki calibration gives a much better results while as the Kalibr failed.

goldbattle, 2022-12-07:
You might want to try the `kalibr_calibrate_cameras` which assumes the camera is global shutter (similar to the ROS camera calibration).

Calibrating with rolling shutter does not mean that an odometry method (like orb_slam) will do better if that method does not take into account that rolling shutter effect. I would actually expect global shutter camera calibration method to perform better on a odometry method which only takes into account a global shutter camera, as compared to using a rolling shutter calibration on that same method. Now if the odometry method took into account the rolling shutter readout time, then this might be another story...

closed 2022-12-08 by zoldaten

Rajni-kanth, 2023-10-26:
hi , How can i use the calibrated file i got from ros wiki tutorial with orb slam @zoldaten