ethz-asl/kalibr #587: can't detect tag with tagcode 36h11

heijieke, 2022-12-12:
camera can't detect tag with tagcode 36h11,but can detect some tag with 16h5.The board is Aprilgrid_6x6_0.8x0.8.Is it because the resolution of camera too low?I try the resolution with 1920*1080 and 640*480, but all failed.I'm wired and it bothering me many days.Plea help! @goldbattle
this is 36h11:
[image]
this is 16h5:
[image]

goldbattle, 2022-12-19:
It shouldn't have any issue on lower resolutions. Do you have a bag that you could share that has this problem? What exact command are you running to get this issue?

heijieke, 2022-12-20:
> It shouldn't have any issue on lower resolutions. Do you have a bag that you could share that has this problem? What exact command are you running to get this issue?
unfortunately,i have deleted my bag,you can test with the picture above.i use the command "rosrun kalibr kalibr_calibrate_cameras --target april_6x6_80x80cm.yaml --topics /camera/color/image_raw --models pinhole-radtan --bag data.bag --show-extraction". And after that, when i change the target to chessbord,it performs normal.it just can't detect AprilGrid.

closed 2022-12-29 by goldbattle