ethz-asl/kalibr #589: Not all images used in intrinsic calibration...

Oo-RR-oO, 2022-12-23:
Why aren't all images used in the calibration intrinsic?
I had taken 86 images, but only 25 images used?
[image]

goldbattle, 2022-12-23:
This is due to outlier filtering. You can disable it and the final outlier
filtering with parameters (see --help).

Oo-RR-oO, 2022-12-24:
> This is due to outlier filtering. You can disable it and the final outlier
> filtering with parameters (see --help).
> 
> 
> > Why aren't all images used in the calibration intrinsic?
> > I had taken 86 images, but only 25 images used?
> > [image: image]
> > <https://user-images.githubusercontent.com/23496607/209256923-c41fbb0b-eb12-4ecc-bbba-9546604cb3cf.png>

All right,thank you

closed 2022-12-24 by Oo-RR-oO

sgraybi, 2025-04-23:
@goldbattle @Oo-RR-oO I'm using the following options: --models ds-none --no-outliers-removal --no-final-filtering --verbose

I also see that not all images are being used in the calibration.  For my dataset "Processed 73 images with 32 images used".  How are these images selected?  Why are some images not being used?  It is also stochastic.  Running the exact same command results in a different number of image being used.

Turning on --verbose some images have "The estimator did not accept this batch" and others "The estimator accepted this batch".  What is the criteria for the estimator accepting or rejecting a batch/image?