ethz-asl/kalibr #592: kalibr_bagextractor bug ?

zoldaten, 2022-12-29:
hi !
i use docker image built with ubuntu 20.04 and kalibr_bagextractor:
`rosrun kalibr kalibr_bagextractor --image-topic /camera --imu-topic /imu --output-folder /data --bag /data/dynamic.bag`
i seems everything i s OK:
```
importing libraries
Extracting 3491 images from topic /camera
      done.                                                          

Extracting 1379 IMU messages from topic /imu
      done.
```
but when i see dir i see imu0.csv and cam0 dir with only 1 image inside - 0000000000.png
that is very strange as when i start calibration  and use --show-exctaction i see all video in bag i recorded.

what is wrong ?

goldbattle, 2022-12-31:
Can you post the bag that has the issue?

zoldaten, 2023-01-05:
sure - https://drive.google.com/file/d/17ZqjSXOwMTgy8eJJLLj7DN3_LUNfr_YO/view?usp=sharing

goldbattle, 2023-02-13:
It looks like the header doesn't have any timestamps, thus it is just overwriting the same image each time.
[image]

I think there should be a way to fall back onto the bag time, but not sure.
If you correct your timestamps, things should perform as expected.

```
rosrun kalibr kalibr_bagextractor --image-topics /camera --output-folder data/. --bag dynamic.bag
```

closed 2023-05-24 by goldbattle