ethz-asl/kalibr #593: Calibration target customization

zpillio, 2023-01-11:
I'm suggesting the following changes in aslam_offline_calibration/kalibr/python/kalibr_create_target_pdf:
- Making calibration target informations (axis and text) optional, because it's not necessary for the calibration and it can save space. I found it really helpful when I experienced with different printing techniques.
- Making it possible to offset the tags. This modification opens the possibility to use more than one calibration targets at the same time.

These are relatively small changes I already did, and before I fork your repo to be able to push it, I wanted to ask you if you are interested in these changes.

ajsnyde, 2023-03-12:
To hijack the feature request, are there any options/plans to have a calibration target in Charuco format? Going to buy a big board of that and was wondering whether kalibr would still be a backup option if I used that board.

Alex-Beh, 2023-03-29:
@zpillio Hi, do you mind to share your experience and lesson learnt on the different printing techniques?

zpillio, 2023-03-29:
> @zpillio Hi, do you mind to share your experience and lesson learnt on the different printing techniques?

For a long time I was using ink printing:
+ they are really accurate and avaiable in large poster sizes
- it uses paper, which is sensitive to humidity
- needs to be glued on top of a flat surface

Laser engraving:
+ you can do it directly on rigid flat surfaces, except metals
- not ideal for large surfaces, because it's slow
- contrast can be problematic: it depends on the material how it darkens, and needs some experiments to find the sweet spot between high contrast and burning it

Screen printing:
+ you can print directly on any rigid flat surfaces
- it has relatively low precision compared to the other two within 1 mm, but noticeable

In my case screen printing on 3mm Aluminum Composite Panels was the best in 0.5m x 0.5m size. These boards are light, rigid, flat and durable.

I would also suggest to ask the printing company to use matte materials or coating. They usually prefer shiny, which can make the lighting difficult.