ethz-asl/kalibr #595: Opimization divered due to bad initialization

Hateley123, 2023-01-25:
I am using Flir black fly S cameras with fisheye lenses and a field of few of 135 degrees. I am able to calibrate both cameras individually with kalibr, but I am unable to calibrate them together and when I do I keep getting the following error. The ds-none models should be fine since I used that same model to calibrate each camera individually, but regardless of the model I use when they are together, I get the same error as listed below. 

rosrun kalibr kalibr_calibrate_cameras --bag /home/ig-handle/2023_01_19_11_35_09/cam1_cam2.bag --topics /F1/image_raw /F2/image_raw --models ds-none ds-none --target /home/ig-handle/catkin_ws/src/ig_handle/target.yaml 
importing libraries
Initializing cam0:
	Camera model:	  ds-none
	Dataset:          /home/ig-handle/2023_01_19_11_35_09/cam1_cam2.bag
	Topic:            /F1/image_raw
	Number of images: 504
Extracting calibration target corners
  Extracted corners for 234 images (of 504 images)                              
	Projection initialized to: [ -0.19759308   0.60839285 429.00181981 428.28929492 656.71148617
 535.27746597]
	Distortion initialized to: []
Initializing cam1:
	Camera model:	  ds-none
	Dataset:          /home/ig-handle/2023_01_19_11_35_09/cam1_cam2.bag
	Topic:            /F2/image_raw
	Number of images: 504
Extracting calibration target corners
  Extracted corners for 206 images (of 504 images)                              
	Projection initialized to: [ -0.17992725   0.60304266 424.91728933 424.79957712 630.70315805
 527.02917432]
	Distortion initialized to: []
initializing initial guesses
	 initializing camera pair (0,1)...  
initialized baseline between cam0 and cam1 to:
[[ 0.99625855 -0.03269369  0.08000012 -0.5909733 ]
 [ 0.07439956  0.79544821 -0.60143733  0.39402985]
 [-0.04397274  0.60513905  0.79490447 56.58452991]
 [ 0.          0.          0.          1.        ]]
initialized cam0 to:
	 projection cam0: [  2.21627654  -6.65034447 430.79643179 437.5897476  584.6956868
 556.47141538]
	 distortion cam0: []
initialized cam1 to:
	 projection cam1: [ -0.98284825  -0.1677062  382.7034409  458.02408625 854.93739033
 451.90269965]
	 distortion cam1: []
initializing calibrator
starting calibration...
[ERROR] [1674681005.632005]: Did not converge in maxIterations... restarting...
[ WARN] [1674681005.633678]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)
[ WARN] [1674681005.635367]: Restarting for a new attempt...
Reinitialize the intrinsics for camera 0
	Projection initialized to: [ -0.19759308   0.60839285 429.00181981 428.28929492 656.71148617
 535.27746597]
	Distortion initialized to: []
Reinitialize the intrinsics for camera 1
	Projection initialized to: [ -0.17992725   0.60304266 424.91728933 424.79957712 630.70315805
 527.02917432]
	Distortion initialized to: []
initializing initial guesses
	 initializing camera pair (0,1)...  
initialized baseline between cam0 and cam1 to:
[[ 0.99812767 -0.03435842  0.05060287 -0.57291649]
 [ 0.05872263  0.76974437 -0.63564555  0.37899912]
 [-0.0171115   0.63742694  0.77032077 49.60837022]
 [ 0.          0.          0.          1.        ]]
initialized cam0 to:
	 projection cam0: [  1.83462078  -4.328681   432.56577653 436.07602205 580.11647265
 533.74657733]
	 distortion cam0: []
initialized cam1 to:
	 projection cam1: [ -0.98040931  -0.15311279 391.15937356 450.88072582 849.84464349
 458.96592532]
	 distortion cam1: []
initializing calibrator
starting calibration...
  Progress 3 / 351 	 Time remaining: 14s                  [ERROR] [1674681311.661959]: Did not converge in maxIterations... restarting...
[ WARN] [1674681311.663850]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)
[ WARN] [1674681311.665624]: Restarting for a new attempt...
Reinitialize the intrinsics for camera 0
	Projection initialized to: [ -0.19759308   0.60839285 429.00181981 428.28929492 656.71148617
 535.27746597]
	Distortion initialized to: []
Reinitialize the intrinsics for camera 1
	Projection initialized to: [ -0.17992725   0.60304266 424.91728933 424.79957712 630.70315805
 527.02917432]
	Distortion initialized to: []
initializing initial guesses
	 initializing camera pair (0,1)...  
initialized baseline between cam0 and cam1 to:
[[ 0.99812767 -0.03435842  0.05060287 -0.57291649]
 [ 0.05872263  0.76974437 -0.63564555  0.37899912]
 [-0.0171115   0.63742694  0.77032077 49.60837022]
 [ 0.          0.          0.          1.        ]]
initialized cam0 to:
	 projection cam0: [  1.83462078  -4.328681   432.56577653 436.07602205 580.11647265
 533.74657733]
	 distortion cam0: []
initialized cam1 to:
	 projection cam1: [ -0.98040931  -0.15311279 391.15937356 450.88072582 849.84464349
 458.96592532]
	 distortion cam1: []
initializing calibrator
starting calibration...
  Progress 15 / 351 	 Time remaining: 1m 3s                  [ERROR] [1674681621.464515]: Did not converge in maxIterations... restarting...
[ WARN] [1674681621.466496]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)
[ERROR] [1674681621.468286]: Max. attemps reached... Giving up...

goldbattle, 2023-02-13:
Do you have the bag file that this happens?
It looks like the transformation is saying they ar 58 meters apart.
```
initialized baseline between cam0 and cam1 to:
[[ 0.99625855 -0.03269369  0.08000012 -0.5909733 ]
 [ 0.07439956  0.79544821 -0.60143733  0.39402985]
 [-0.04397274  0.60513905  0.79490447 56.58452991]
 [ 0.          0.          0.          1.        ]]
```

Are you sure your calibration target has correctly configured units?

closed 2023-02-13 by Hateley123

goldbattle, 2023-02-14:
Cross ref https://github.com/ethz-asl/kalibr/issues/596#issuecomment-1428791493