ethz-asl/kalibr #599: Translation error (near to 0) between IMU and camera.

LarryDong, 2023-03-19:
The translation between my IMU and camera is about 10cm. 
However, the kalibr gives the wrong translation (x,y,z are all near to 0) but the rotation seems correct. 
```
T_ci:  (imu0 to cam0): 
[[ 0.98235183  0.18703993  0.0009721   0.00003113]
 [ 0.01204317 -0.05806375 -0.99824023  0.00008589]
 [-0.18665434  0.98063483 -0.05929158  0.00000564]
 [ 0.          0.          0.          1.        ]]
```
Here, the last column for translation is wrong. 
Here is the results pdf: [test-report-imucam.pdf](https://github.com/ethz-asl/kalibr/files/11011297/test-report-imucam.pdf)

What I've found instersting is that, **the estimated bias of acc and gyro is always constant** (the same as mentioned issues below):

[image]

---
I've viewed some of the issues, and found some others have the similar issues, such as [https://github.com/ethz-asl/kalibr/issues/572#issue-1397895936](https://github.com/ethz-asl/kalibr/issues/572#issue-1397895936), [https://github.com/ethz-asl/kalibr/issues/451#issuecomment-964985577](https://github.com/ethz-asl/kalibr/issues/451#issuecomment-964985577).
My platform: ubuntu 18.04, ros melodic.
I've changed IMU frequency from 100Hz to 1000Hz, and I also tried camera's fps at 4, 10, and 20, but they didn't help. Also, I changed the translation between camera and IMU, 3cm to 10cm, in either imu's z axis or imu's y axis.

The camera's calibration is given by `kalibr` and the matlab gives the similar results. Thre reprojection error is lower than 1px.
The IMU is KVH1750 (about $ 30,000) and calibrated by [https://github.com/gaowenliang/imu_utils](https://github.com/gaowenliang/imu_utils).
When recording the rosbag, 3-axis rotation and translation are smooth and slow, and 3-axis are well excited I believe.

Hope to get your help.

goldbattle, 2023-03-19:
Does the accelerometer have gravity in it?
Looking at the IMU timestamps, there might be something wrong there.
[image]

Why are some sequential IMU readings have 30ms, while others have near zero?
This will need to be addressed (should be one flat light at 1/1000 or 1/100).

The accelemeter noises seem a bit large here, so maybe there is something there too?
[image]

Have you tried processing your static IMU bag with:
https://github.com/ori-drs/allan_variance_ros

How long is your static bag?
Are you able to post any of the bag files you collected?

LarryDong, 2023-03-20:
@goldbattle Thanks for your quick reply. 
> Does the accelerometer have gravity in it?   

Yes.
> Why are some sequential IMU readings have 30ms, while others have near zero?   

Yes it is an issue indeed. Thanks for point it out!  
I'm using `boost::asio::serial_port` to read IMU from `ttyUSB0`, and I found that there is a 30ms delay (I got 3 sequences every 30ms).   
I have solved this timestamp issue by chaging the `latency_timer` to 1 (default: 16) of serial_port on linux system, by:
```
sudo sh -c 'echo 1 > /sys/bus/usb-serial/devices/ttyUSB0/latency_timer'
```
See [This](https://stackoverflow.com/questions/13126138/low-latency-serial-communication-on-linux) Stackoverflow. Hope it can help others who have the same problem.

> The accelemeter noises seem a bit large here, so maybe there is something there too?
>  Have you tried processing your static IMU bag with: [allan_variance_ros](https://github.com/ori-drs/allan_variance_ros)?

I will try it soon. Thanks.

> How long is your static bag?

about 2 hours. I will use a longer one later.

> Are you able to post any of the bag files you collected?

If there are still issues, I will upload one.

Thanks again for your reply.  Please keep this issue open, and I will close it if there's no other issues.

LarryDong, 2023-03-21:
> The accelemeter noises seem a bit large here, so maybe there is something there too?

Yep another issue found in gyroscope. The unit should be rad/s but I'm using degree.  
The calibration results seems correct (although not so accurate). 
I appreciate your suggestions. @goldbattle 

In summary, two reasons lead to this issue:
- The latency due to the serial port, leading to fluctuating time_offset between camera and imu;
- The unit of gyroscope, rad but not degree.

closed 2023-03-21 by LarryDong

Jawaheru, 2024-04-26:
Where did u change the gyroscope unit?

LarryDong, 2024-04-26:
> Where did u change the gyroscope unit?

Just multiply the value by "180/PI" when publishing, or using a new node to subcribe and re-publish the value.

Jawaheru, 2024-04-26:
Yes, but the gyro data we enter for the calibration is only the random walk
and white noise.
You mean the recorded imu data shud be captured in rad right?