ethz-asl/kalibr #605: kalibr_create_target_pdf command not working in docker environment.

Ruthrash, 2023-03-31:
I have built the docker container using commands [here](https://github.com/ethz-asl/kalibr/wiki/installation#:~:text=docker%20build%20%2Dt%20kalibr%20%2Df%20Dockerfile_ros1_20_04%20.%20%23%20change%20this%20to%20whatever%20ubuntu%20version%20you%20want). However, when I try to run, 

```kalibr_create_target_pdf  --h
```

inside the docker container, I get command not found error

goldbattle, 2023-03-31:
Run it with rosrun. It is in the "kalibr"  package.

Ruthrash, 2023-03-31:
Ah that makes sense. My bad, the instructions [here](https://github.com/ethz-asl/kalibr/wiki/calibration-targets#:~:text=kalibr_create_target_pdf%20%2D%2Dh)  was a bit unclear. 

Thanks a lot!