ethz-asl/kalibr #607: [ERROR] [1681013049.615084]: argument --imu is required

YOUZHUYOU, 2023-04-09:
I installed the kalibr, when i run a sample using the commends:
`kalibr_calibrate_imu_camera --target /home/zy/kalibr_workspace/dynamic/ april_6x6.yaml --cam /home/zy/kalibr_workspace/dynamic /camchain.yaml --imu/home/zy/kalibr_workspace/dynamic /imu_adis16448.yaml --bag /home/zy/kalibr_workspace/dynamic /dynamic.bag --bag-from-to 5 45
`
the error came out like that:
`[ERROR] [1681013049.615084]: argument --imu is required`
Why? And how can i solve it?

goldbattle, 2023-04-09:
Read the --help dialog. You need to specify the IMU model you wish to run
with. For example --imu calibrated

YOUZHUYOU, 2023-04-09:
Do you mean i correct the commend like
 `--imu calibrated/home/zy/kalibr_workspace/dynamic /imu_adis16448.yaml`

YOUZHUYOU, 2023-04-09:
Do you mean to correct the commend like:
`--imu  calibrated/home/zy/kalibr_workspace/dynamic /imu_adis16448.yaml `
I tried it, then another error came:
`[ERROR] [1681016189.943579]: unrecognized arguments: april_6x6.yaml camchain.yaml dynamic.bag
`

closed 2023-04-09 by YOUZHUYOU

goldbattle, 2023-04-10:
This is the example command from the wiki:
https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration
```
rosrun kalibr kalibr_calibrate_imu_camera \
	--target april_6x6.yaml \
	--imu imu_adis16448.yaml \
	--imu-models calibrated \
	--cam cam_april-camchain.yaml \
	--bag imu_april.bag
```

You can also run the following and read the help dialog.
```
rosrun kalibr kalibr_calibrate_imu_camera --help
```

Not sure if you mean to have a "space" in your directory path, but if this is truly the case you need to probably quote the string (standard linux stuff here).