ethz-asl/kalibr #608: Incorrect checkerboard pattern detection

shrutarv, 2023-04-14:
I have used multiple combinations of checkerboard patterns for calibration. But, kalibr does not detect the corners in an ordered way. please have a look at the attached image. Is this just a visualization error? Please help me resolve this.
[image]
[image]

goldbattle, 2023-04-14:
No, this is the problem of using checkerboards (it isn't clear the ordering so you can have it "flip" mid way through a dataset). This is also why it is not possible to have a square checkerboard. See the top of the wiki page:
https://github.com/ethz-asl/kalibr/wiki/calibration-targets

shrutarv, 2023-04-14:
I tried with april grid. But, kalibr does not detect the grid patterns. Please have a look at the screenshot
[image]

goldbattle, 2023-04-14:
Is this a thermal camera? The contrast seems very low. Likely it is a resolution or focus problem with your sensor.

shrutarv, 2023-04-17:
> No, this is the problem of using checkerboards (it isn't clear the ordering so you can have it "flip" mid way through a dataset). This is also why it is not possible to have a square checkerboard

I also tried with a 6x9 checkerboard pattern. Please have a look at the second image. It looks like there is a bug in kalibr. Could you please hep me out here.

goldbattle, 2023-04-17:
The detection of the checkerboards look ok in your original posting. What is the problem or could you clarify?

See the example here:
https://docs.opencv.org/3.4/d4/d94/tutorial_camera_calibration.html
[image]

Maybe you are referring to the colors not being per-row in your picture? This maybe is a bug in visualization or how the extractions are tracked. Is this what you mean / could you clarify? I originally thought you were talking about that detection can be on the horizontal rows or the vertical columns.

goldbattle, 2023-05-24:
Very good point, should be looked into.