ethz-asl/kalibr #609: Why is the number of rows and columns of april grid are limited to a minimum of 3

Ruthrash, 2023-04-17:
https://github.com/ethz-asl/kalibr/blob/f581b27b68051200c1786e5f901cfe059e3f93b2/aslam_offline_calibration/kalibr/python/kalibr_common/ConfigReader.py#L613 

Any prinicipled reason for limitied minimum number of rows and columns to 3?

goldbattle, 2023-04-17:
I believe this is because it is detecting the inner corners of the checkerboard, which is not possible with less than 3 rows or columns. Maybe you could search about OpenCV's checkerboard detection to understand more.

Ruthrash, 2023-04-17:
I understand that's the case for the checkerboard, but my question was for the april grid. I can think of no reason why we need to limit it for the april grid case

goldbattle, 2023-07-20:
Very true, would need to test to see if that gives good results.