ethz-asl/kalibr #611: d435 extrinsic estimation with April Board

Deok-min, 2023-04-24:
Hi, thank you for sharing great work.
By the way, I'm trying to estimate Transformation matrix between April board and wrist camera (rs d435).
But, the visualized result is not accurate. The projected depth image and estimated board frame is not aligned. Could you give me some advice to improve the accuracy?
[image]

And my code followed below steps.
1. First i measured intrinsic parameters provided code "kalibr_calibrate_rs_cameras".
2. Generate Camera.geometry using the intrinsic parameters obtained from above step.
[image]

3. Estimate Extrinsic parameters using provided function, '_generateExtrinsicsInitialGuess()'.
[image]

How can I improve the tf accuracy? Thank you for reading.

goldbattle, 2023-05-24:
Maybe there is an additional transformation to your depth camera. Not sure.

goldbattle, 2023-05-24:
Feel free to reopen if this is an issue with the Kalibr tools and not your implementation.

closed 2023-05-24 by goldbattle