ethz-asl/kalibr #612: Wrong camera-Imu translation estimation on a ground vehicle (car) using Kalibr.

Vmine667, 2023-05-11:
Hello, 
I'm calibrating the mono-camera (25fps) and IMU (100Hz) on a ground vehicle (car) using Kalibr. While my rotation matrix estimation is relatively accurate, the translation between the IMU and camera is completely erroneous like 2-3 cm instead of about 1m. What steps can I take to rectify this issue? 
PS : The estimation of the imu random-walk and noise was found by using allaallan_variance_ro after 6h of recording.
[image]

Command line I use : `rosrun kalibr kalibr_calibrate_imu_camera --imu imu.yaml --cam cam_april_camchain --bag my.bag --target aprel_6x6.yaml`

yamls : [https://drive.google.com/drive/folders/1uh1k35E3g4jKGLw-5aM5bKM9wmeucVRU?usp=sharing](url)

My Rosbag : [https://drive.google.com/drive/folders/1PomVCSV-uVHh7ZrMcSuWbBDDWGQKU3WH?usp=sharing](url)

Pdf_report : [https://drive.google.com/drive/folders/1YVESCDCmkVr15Efl_F8DsGCIHDUzKvKh?usp=sharing](url)

Thank you,

goldbattle, 2023-05-11:
Please use the latest master code and try to regenerate.
Ensure that you have full 6dof motion (i.e. you can't simply record a
dataset of a vehicle moving in front of a board).

Vmine667, 2023-05-15:
Thank you for your quick response, since I'm trying to calibrate a ground vehicle, I can't do a  rotation around y and x axis (roll and pitch). Can you suggest any alternative?

goldbattle, 2023-05-15:
I don't have anything besides making the sensor rig removable, or somehow
making the ground vehicle move more (ramp on one wheel / jack?).

closed 2023-05-24 by goldbattle