ethz-asl/kalibr #616: Calibration data from kalibr

shrutarv, 2023-05-17:
Hello,

I have successfully calibrated a stereo setup of a DVXplorer event camera and a RGB camera using kalibr. I have a point in the event camera frame and I would like to get the corresponding point in the RGB camera frame. How to get that using the data obtained from calibration using the kalibr?

goldbattle, 2023-05-17:
Kalibr does not support event cameras.

closed 2023-05-24 by goldbattle

JiaqiangZhang, 2025-01-17:
Hi, @shrutarv thank you for the issue.

I am working on calibrating a DVXplorer event camera stereo setup as well. Did you use the DV-GUI to calibrate, or some other tool? I followed the [official instructions](https://docs.inivation.com/software/dv/gui/calibrate-event-camera.html) to calibrate the stereo setup, but it keeps failing with high errors. 

[image]