ethz-asl/kalibr #621: Question: Deriving projection matrix (ROS camera info) using extracted kalibr yaml file

arjay55, 2023-06-05:
Hi,

I cannot directly find any projection matrix in the Kalibr yaml file. I wanted to translate this file for use in ROS 2 camera info, but I am lacking the matrix for correct stereo calibration.

Example:
```
projection
856.167319 0.000000 197.679413 -178.286885
0.000000 856.167319 379.488960 0.000000
0.000000 0.000000 1.000000 0.000000
```
Need help on how to derive such parameters, so that I can obtain the correct point cloud scale. Thanks.

goldbattle, 2023-06-08:
Creating this isn't super straightforward since there is distortion. Typically you only have this matrix after stereo rectification. These opencv documentation should help you I think:
- Overview: https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#details
- stereoCalibrate(): https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga91018d80e2a93ade37539f01e6f07de5
- stereoRectify(): https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6

I think you can call stereoRectify() and request a specific projection matrix for both cameras, and then from there compute the distortion maps to then correct your incoming images.

closed 2023-07-20 by goldbattle

zoldaten, 2023-08-23:
@goldbattle could tell please,
if i have simple mono camera my matrixs will be looking like that?

camera matrix
351.774060 0.000000 290.713492
0.000000 468.945113 235.782768
0.000000 0.000000 1.000000

distortion
0.11229715732461544 -0.15069645913928714 0.0014552750357630418 0.00088025520757 0.000000

rectification
0.0 0.0 0.0
0.0 0.0 0.0
0.0 0.0 0.0

projection
0.0 0.000000 0.0 0.000000
0.000000 0.0 0.0 0.000000
0.000000 0.000000 0.000000 0.000000

according https://stackoverflow.com/questions/16101747/how-can-i-get-the-camera-projection-matrix-out-of-calibratecamera-return-value
i do not quite understand what to multiply with camera matrix.

my Kalibr:
cam0:
  camera_model: pinhole
  distortion_coeffs: [0.11229715732461544, -0.15069645913928714, 0.0014552750357630418, 0.0008802552075759202]
  distortion_model: radtan
  intrinsics: [351.7740605661654, 468.9451131575785, 290.71349241118145, 235.78276791924256]
  line_delay: 6.508060238881519e-05
  resolution: [640, 480]
  rostopic: /image_raw

goldbattle, 2023-08-26:
Kalibr has:
```
intrinsics: [351.7740605661654, 468.9451131575785, 290.71349241118145, 235.78276791924256]
intrinsics: [ fx fy cx cy
```

Translates to:
```
K
351.774060 0.000000 290.713492
0.000000 468.945113 235.782768
0.000000 0.000000 1.000000

K
fx 0.000000 cx
0.000000 fy cy
0.000000 0.000000 1.000000
```

Distortion for radtan should be 1-to-1 ignoring the 5th element which is considered to be zero in kalibr.
e.g. only k1, k2, and p1, p2 are consider as compared to this more complete model in opencv
https://docs.opencv.org/3.4/d9/d0c/group__calib3d.html#details
[image]