ethz-asl/kalibr #626: Question about the Detection failed.

hanseongbugi, 2023-06-25:
I run kalib_calibrate_imu_camera with verbose option.
It printed out the phrase detection failed as a result of execution.
I made the bag file that I needed to run by myself through a mobile device.
Looking at the results output by the verbose option, it was found that the image was blurred.
I think this happened because it didn't detect the edge well even though it used a Gaussian filter.
The video collection was conducted by referring to the YouTube video.
Please let me know what kind of solution there is

goldbattle, 2023-06-25:
You need to have good focus / sharpness to extract. Could you post a bad / picture of the problem? Additionally, you need to first perform camera calibration before IMU-camera, which will also require good quality images.

hanseongbugi, 2023-06-26:
[https://drive.google.com/file/d/1pcRNWEPAZxH2bazmh4uzkb1RjnkuoQho/view?usp=sharing](url) here is my bag file

goldbattle, 2023-06-26:
This looks like a case of either using a rolling shutter camera and having high motion blur. It is able to extract on still frames, but has a hard time while moving. I recommend increasing the shutter speed.
[image]

hanseongbugi, 2023-06-27:
Currently my camera gets frames every 5ms, imu gets data for 10ms. 
Also, my mobile device, Samsung Note 5, has a limit of 5ms of camera shutter speed.
Can you tell me if the value of 5ms is too fast or too slow to make the right result in kalibr

goldbattle, 2023-09-14:
33ms should be fine (it is 30hz) the problem I think is the visual rolling shutter you can see. If you are able to record at a higher framerate then this could go away due to the lower exposure time.

closed 2023-12-31 by goldbattle