ethz-asl/kalibr #630: [Error]libsm_python.so: undefined symbol: _ZN5boost6python15instance_holder8allocateEP7_objectmmm

myself-wutong, 2023-07-10:
hello，I am attempting to install this package on ubuntu20.04.  But I encountered an error during runtime：
/home/work/kalibr_workspace/devel/lib/python3/dist-packages/sm/libsm_python.so: undefined symbol: _ZN5boost6python15instance_holder8allocateEP7_objectmmm

This link states that this is caused by cmake not finding the correct Python： https://stackoverflow.com/questions/70582192/boostpython-undefined-symbol-zn5boost6python15instance-holder8allocateep7-ob
I tried to solve the problem using this method but failed. Is there a solution to the problem

goldbattle, 2023-07-10:
This is likely a python version issue, so I would check what version of
python you have `python --version` and see what one the cmake is building
against.
If you can't get it to compile, you can always run the docker.

myself-wutong, 2023-07-11:
> This is likely a python version issue, so I would check what version of python you have `python --version` and see what one the cmake is building against. If you can't get it to compile, you can always run the docker.
> […](#)

Thank you, I have reconfigured a new virtual machine environment and only configured the required dependencies.  The compilation and operation on the virtual machine are very smooth, and I think it should be caused by some version conflicts between Python and CMake

closed 2023-07-20 by goldbattle