ethz-asl/kalibr #633: calibration of non synchronized cameras

lsorgi, 2023-07-17:
Hello! 

I am having issues calibrating a system with 2 non-synchronized cameras with divergent optical axis, i.e. with only ~30% of the field of view overlap.

If I run the intrinsic calibration of both cameras independently it works correctly, undistorsion look good, metrics good, all good. 
Commands for intrinsic calibration is

```
rosrun kalibr kalibr_calibrate_cameras  --bag <KALIBR_BAGFILE>  --topics /cam0/image_raw  --models pinhole-equi  --target <TARGET_YAML>
rosrun kalibr kalibr_calibrate_cameras  --bag <KALIBR_BAGFILE>  --topics /cam1/image_raw  --models pinhole-equi  --target <TARGET_YAML>
```

If I run the calibration of both cameras simultaneously it fails, not being able to correctly initialize the stereo pairs.  
The command for stereo calibration is

```
rosrun kalibr kalibr_calibrate_cameras  --bag <KALIBR_BAGFILE>  --topics /cam0/image_raw --topics /cam1/image_raw  --models pinhole-equi pinhole-equi  --target <TARGET_YAML>
```

I naively assumed that the temporal offset between the cameras was handled by the tool... is it not the case? do the 2 camera stream need to be strictly synchronized?

kololheUSL, 2023-07-17:
I faced a similar issue. What you can do instead is provide a measure of the asynchronization. AFAIK, Kalibr does do approximate synchronization if the topics are not synchronized. 

You can pass the tolerance for approximate synchronization:
`
--approx-sync MAX_DELTA_APPROXSYNC
                        Time tolerance for approximate image synchronization
                        [s] (default: 0.02)
`

To identify the time difference, you can use `rostopic delay` which will give you delays between sequential ROS messages for a topic (NOT between multiple topics)

You can also increase the number of iterations, or include more images of your target in the overlap of the two cameras.

closed 2023-07-18 by lsorgi

sun421-cloud, 2024-08-01:
> thx!

Hello, have you successfully calibrated the unsynchronized stereo camera with the above method? Or hardware alignment was eventually adopted? 
Looking at your camera model, do you use a fisheye camera?

lsorgi, 2024-08-03:
Hi! I use fisheye camera models and we eventually synchronized the cameras