ethz-asl/kalibr #635: Did not converge. Bad distortion values.

bhushanap, 2023-07-20:
I am really struggling to get low distortion values for even a single camera calibration. I wanted to run stereo cameras but for some reason the values are extremely weird.

Earlier I was trying both cameras at once, and that failed. So I started to do mono camera to see if that was the issue. And indeed each of the camera projections is bad. I am using a standard April 6x6 but a slightly changed [yaml](https://gtvault-my.sharepoint.com/:u:/g/personal/bpawaskar3_gatech_edu/EWnrs5GZ8zhDjeHpmFnCDTcBDoiVnFv2ehLtOSSNxv1ovw?e=vUlFhT) file because of some printing issue. I have tried multiple both equi and radtan but to no avail. I tried initializing the focus values to multiple values now. I have also tried making multiple bag files, mono, color, raw, compressed but none of them seem to solve the distortions issue.

Here's the log of the output I get for various initializations of focal length for pinhole-equi on raw camera data:

```
rosrun kalibr kalibr_calibrate_cameras --target april_6x6.yaml --models pinhole-equi --topics /right_camera/image_raw --bag test1.bag --bag-freq 10.0 --show-extraction
importing libraries
Initializing cam0:
	Camera model:	  pinhole-equi
	Dataset:          test1.bag
	Topic:            /right_camera/image_raw
[ WARN] [1689843141.641607]: BagImageDatasetReader: truncated 933 / 1191 images (frequency)
	Number of images: 258
Extracting calibration target corners
  Extracted corners for 258 images (of 258 images)                              
Initialization of focal length failed. Provide manual initialization: 
4.5
Initializing focal length to 4.5
	Projection initialized to: [  11.67501038   -8.00262905  582.110891    440.06002513]
	Distortion initialized to: [-12.13285292   8.38860694  -4.33760376   1.09703222]
initializing initial guesses
initialized cam0 to:
	 projection cam0: [  11.67501038   -8.00262905  582.110891    440.06002513]
	 distortion cam0: [-12.13285292   8.38860694  -4.33760376   1.09703222]
initializing calibrator
starting calibration...
[ERROR] [1689843212.386015]: Did not converge in maxIterations... restarting...
[ WARN] [1689843212.389846]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)
[ WARN] [1689843212.392080]: Restarting for a new attempt...
Reinitialize the intrinsics for camera 0
Initialization of focal length failed. Provide manual initialization: 
25
Initializing focal length to 25
	Projection initialized to: [   8.60262198   10.6169384   584.33074979  438.93433131]
	Distortion initialized to: [ 6.06408831  3.65666951 -2.88065844  0.20502104]
initializing initial guesses
initialized cam0 to:
	 projection cam0: [   8.60262198   10.6169384   584.33074979  438.93433131]
	 distortion cam0: [ 6.06408831  3.65666951 -2.88065844  0.20502104]
initializing calibrator
starting calibration...
[ERROR] [1689843222.760098]: Did not converge in maxIterations... restarting...
[ WARN] [1689843222.763415]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)
[ WARN] [1689843222.765348]: Restarting for a new attempt...
Reinitialize the intrinsics for camera 0
Initialization of focal length failed. Provide manual initialization: 
400
Initializing focal length to 400
	Projection initialized to: [   5.77254913    5.46862403  581.01493482  559.58860725]
	Distortion initialized to: [ 268.34114458 -277.06581862    7.55495588   84.63046897]
initializing initial guesses
initialized cam0 to:
	 projection cam0: [   5.77254913    5.46862403  581.01493482  559.58860725]
	 distortion cam0: [ 268.34114458 -277.06581862    7.55495588   84.63046897]
initializing calibrator
starting calibration...
[ERROR] [1689843242.249279]: Did not converge in maxIterations... restarting...
[ WARN] [1689843242.253547]: Optimization diverged possibly due to a bad initialization. (Do the models fit the lenses well?)
[ERROR] [1689843242.257446]: Max. attemps reached... Giving up...
```

[image]

The [cameras](https://www.edmundoptics.com/p/flea3-fl3-u3-13e4c-c-11.8-color-usb-3.0-camera/29805/) and [lenses](https://www.edmundoptics.com/p/45mm-c-series-fixed-focal-length-lens/29328/) used are for pointgray cameras.

This is the static [bag file](https://gtvault-my.sharepoint.com/:u:/g/personal/bpawaskar3_gatech_edu/EeNLQ87XgCdCk_GwcxenZysBfCXaM53H9JKIB-tNc1oW1g?e=K8BLnq) I have been using for calibration. As you can see, the tags are detected well.

Any ideas for how I can improve calibration?

goldbattle, 2023-08-26:
I think the board is moved in the same plane, so you likely need to "pitch" the whole whiteboard forward and backwards. Otherwise the dataset looks ok besides not exciting this board in all 6dof motion. Additionally, it is only in the middle of the image, but you should ensure to move it to all regions of the FOV. Either you can move the camera and keep the board static, or move the board in full 6dof motion (full rotation, scale, translation, and skew in the FOV of the camera).

goldbattle, 2023-08-26:
The full FOV is particularly important as detections near the edge of the FOV are key to accurate distortion estimation.

closed 2023-09-14 by goldbattle