ethz-asl/kalibr #636: Calibration of 4 cameras facing front, left, right, and back, all pointed outwards from a central point

vichan7, 2023-07-20:
Hi! I want to calibrate 4 cameras attached to a robot as described in the title. How do you suggest going about this? Would it be possible to use multiple rosbag files (one for each pair of cameras)? Or would it be better to collect one rosbag where the chessboard is rotated around the robot?

Atohme, 2023-08-25:
Hi,
I would suggest you get the parameters of each camera, i.e. distortion and intrinsic.
Then calibrate each camera with your IMU (independently). This way you will have every transformation of each camera to your imu. then you can enter these parameters in the configuration file.
Hope this helps.

goldbattle, 2023-08-25:
Kalibr requires overlapping FOV, so in this case what Atohme has suggested
is the only way to go about calibrating the relative transformation between
each (e.g. relate them all through the IMU).

closed 2023-09-14 by goldbattle