ethz-asl/kalibr #640: Numpy_eigen.libnumpy_eigen not found

Aishkrish18, 2023-08-14:
[image]

[image]

Although the repository gets built, i keep gettingthis error from numpy_eigen. I really dont know how to solve this, would be better if someone could help me with this so that I can perform calibration. Was attempting to try this for calibrating for OKVIS algorithm but yeah now it entirely fails to work.

Thanks,
Aishwarya

goldbattle, 2023-10-28:
Hi, could you please provide more information about your install? Is this in the docker container?

goldbattle, 2024-01-16:
Please reopen with details to reproduce.

closed 2024-01-16 by goldbattle