ethz-asl/kalibr #641: Gyro - Accelerometer units

ManChrys, 2023-08-25:
Hello,

i have read a lot of issues about the units of gyroscope and accelerometer and there are different opinions about that. 

What's the correct units ? 

Thanks in advance.

goldbattle, 2023-08-25:
Accel: m/s2
Gyro: rad/s

See the standard: https://www.ros.org/reps/rep-0145.html

closed 2023-08-26 by goldbattle

ManChrys, 2023-08-28:
Hello @goldbattle,

thanks for the response,
yeah i know that generally these are the units. But for kalibr i saw this table

[image]

goldbattle, 2023-08-28:
These are for the noise parameters. The units for them are in the last
column.

ManChrys, 2023-08-28:
yeah i was asking about noise params which i have to fill in yaml file. Its my mistake that i didnt specified.

These units in wiki documentation are the correct right ?

goldbattle, 2023-08-28:
Yes!