ethz-asl/kalibr #643: bagcreator or record directly a ros bag ?

mahammadirfan, 2023-09-07:
Hi,
I want to calibrate stereo-inertial, and have few questions:
1) I have a ROS installed system with the sensors, Do i need to start the bagcreator and convert it into rosbag file or i can directly record everything in a rosbag file ?

Thanks!

mahammadirfan, 2023-09-07:
I have one more question to ask :  
2) I am calibrating dji m300 drone.....it has inbuilt stereo cameras...for now i am using just front....with its own IMU....do i have to check and set IMU + camera cordinate system to any specific for kalibr ?

Thanks!

goldbattle, 2023-09-07:
The wiki describes how to do this. If you can directly record into a rosbag
then there is no need to convert image data into a rosbag data.
You need to ensure they are right handed coordinate systems and use the
correct ROS standard units for the IMU (rad/s and m/s^2).

closed 2023-09-14 by goldbattle