ethz-asl/kalibr #644: [Error] ubuntu 20.04-noetic build failes

KMerdan, 2023-09-10:
I attempted to build [Dockerfile_ros1_20_04](https://github.com/ethz-asl/kalibr/blob/master/Dockerfile_ros1_20_04) locally using the command `docker build -f Dockerfile_ros1_20_04 .`
However, I encountered a failure with the following error:

```bash
dpkg: dependency problems prevent configuration of libpam-fprintd:amd64
```

To resolve this issue, I had to include the following two lines before installing the dependencies:

```Dockerfile
RUN apt-get update && apt-get install libfprint* fprintd* libpam-fprintd* -y 
RUN dpkg --configure -a
```

I'll submit a pull request (PR) to address this problem so that it can assist  whom might face the same issue in the future.

goldbattle, 2023-10-28:
I am unable to reproduce this error. Is this still something you see? What OS version are you building the docker on? I don't see why CI should pass, while for you it doesn't.

goldbattle, 2024-01-16:
I am unable to reproduce locally, feel free to reopen with steps to reproduce it.

closed 2024-01-16 by goldbattle