ethz-asl/kalibr #649: Can the IMU spline interpolation work with < 90Hz

geoeo, 2023-10-09:
I have some pretty bad drivers for my IMU and im getting around 45Hz. I actually calibrated with ~90Hz, but some faulty measurements. This was fine. But now with 45Hz and better measurements Im having problems estimating the translation part of the transformation. Its close to 0. Orientation is fine.  My camera is at 20FPS. So I was wondering if a frequency of around 45Hz is just to low for the spline construction to happen properly

goldbattle, 2023-10-28:
Do you have a bag file with this? I am not sure if the spline construction will be the problem, or the actual optimization. You might need to decrease the knot creation frequency to ensure stability.

geoeo, 2023-10-31:
I double checked the report. It seems the spline interpolation works. But the error is all 0. So your right. It might be in the optimizer. Where can I decrease the knot creation / change the optimizer settings? Are they exposed as flags? Ill try to get you all the files.

In the meanwhile I can upload the report.
[20_fps_45hz_2023-10-09-15-18-42-report-imucam.pdf](https://github.com/ethz-asl/kalibr/files/13218208/20_fps_45hz_2023-10-09-15-18-42-report-imucam.pdf)

xiejianshan, 2024-01-05:
> I have some pretty bad drivers for my IMU and im getting around 45Hz. I actually calibrated with ~90Hz, but some faulty measurements. This was fine. But now with 45Hz and better measurements Im [...]

I meet the same problem with the 50Hz imu data and 30FPS image. But the translation part is fine for another 200Hz imu. Do you solve it? How can I calibrate the low Frequency imu and camera?

geoeo, 2024-01-05:
I upgraded to a better IMU