ethz-asl/kalibr #650: Some errors while catkin_make -DCMAKE_BUILD_TYPE=Release -j2

Tan-B24, 2023-10-11:
catkin_ws/src/kalibr/aslam_nonparametric_estimation/bsplines_python/src/BSplinePython.cpp:40:108: error: ‘_1’ was not declared in this scope
   40 |  return boost::python::make_function(boost::bind(&BiFunction<T>::getBi, BiFunction<T>(bs->getBiVector(t)), _1)
      |                                                                                                                                                                                  ^~

goldbattle, 2023-10-28:
Please provide more details and also ensure to try the docker container.

closed 2024-01-16 by goldbattle

YechenZhou, 2024-05-03:
This link https://blog.csdn.net/nigtkk/article/details/137998334     works for me.