ethz-asl/kalibr #652: Nothing happens when run kalibr_camera_validator

donaldduckkkkk, 2023-11-29:
I haved run three commands:
terminal1 : roscore
terminal2: rosbag play -l mybag.bag
terminal3:
`rosrun kalibr kalibr_camera_validator --cam /home/zy-trl/Downloads/kalibr_validator/dataset_sun7-camchain.yaml --target /home/zy-trl/Downloads/kalibr_validator/target.yaml
`
but nothing happens. "The Camera:" windows and the"Rectified view"windows are all empty. It seems like that there are no image messages to pass in. When i print log in "    def synchronizedCallback(self, *cam_msgs):
" , nothing happens. It seems like this function is not called.
 So what should I do?

goldbattle, 2023-12-31:
Hmm, likely the topics specified are not correct.
Could you provide a bag, the config, and the target you are using so I can reproduce locally?

rytse, 2024-03-01:
I am experiencing something similar using the `Dockerfile_ros1_20_04` image. Specifically, I get these floating error messages:

[image]

**Steps to reproduce:**

_Terminal 1_
```
(host) $ FOLDER=~/Documents/kalibr_validation
(host) $ xhost +local:root
(host) $ docker run -it -e "DISPLAY" -e "QT_X11_NO_MITSHM=1" -v "/tmp/.X11-unix:/tmp/.X11-unix:rw" -v "$FOLDER:/data" kalibr
(container) $ source devel/setup.bash 
(container) $ roscore
```
_Terminal 2:_
```
(host) $ docker exec -it <name> /bin/bash
(container) $ cd catkin_ws/
(container) $ source devel/setup.bash
(container) $ DATAPATH=<path>
(container) $ rosbag play -l ${DATAPATH}/kalibr/kalibr/rosbag/rosbag.bag
```
_Terminal 3:_
```
(host) $ docker exec -it <name> /bin/bash
(container) $ cd catkin_ws/
(container) $ source devel/setup.bash
(container) $ DATAPATH=<path>
(container) $ rosrun kalibr kalibr_camera_validator --cam ${DATAPATH}/kalibr/output/rosbag-camchain.yaml --target ${DATAPATH}/kalibr/output/board.yaml --verbose
```

I've provided `rosbag.bag`, `rosbag-camchain.yaml`, and `board.yaml` in the archive file at [this Google drive link](https://drive.google.com/file/d/1EdcHLgm6rdVpElwxzYW2Xk-ycYrUgAly/view?usp=sharing).

arjunskumar, 2025-04-16:
```
[DEBUG] [1744780634.946308]: Subpix refinement failed for point: 10 with displacement: 1.35984(point removed) 

[DEBUG] [1744780634.946670]: removed target point with reprojection error of 0.772398 (mean: 0.207086, std: 0.114077)

[DEBUG] [1744780634.946677]: removed target point with reprojection error of 0.466178 (mean: 0.207086, std: 0.114077)

[DEBUG] [1744780634.946682]: removed target point with reprojection error of 0.565614 (mean: 0.207086, std: 0.114077)

[DEBUG] [1744780634.946685]: removed target point with reprojection error of 0.465364 (mean: 0.207086, std: 0.114077)

[DEBUG] [1744780634.946690]: removed 4 of 135 calibration target corner outliers

```

@rytse are you to solve this?