ethz-asl/kalibr #654: Some question about "Bad IMU Rate".

liaozihang, 2023-12-01:
In the wiki, there is a mention about [bad IMU timestamps](https://github.com/ethz-asl/kalibr/pull/582) ,
**I have a question about the impact of  bad timestamps**.

Can anyone explain me，a big question is what is the impact of a bad imu rate on vio, which seems to have no relevant explanation found on the wiki or issue.

I would be thankful for any answer.

Thanks

goldbattle, 2023-12-31:
I think this all depends on the noise and the integration methods used.
You want to have a reasonable rate ~100-500Hz IMU with low noise.

The problem with "bad timestamps" is that they are inconsistent, i.e. it is unknow what the true measurement timestamp of each IMU reading is. In this case, I believe basically all VIO / inertial method will fail.

liaozihang, 2024-01-03:
> I think this all depends on the noise and the integration methods used. You want to have a reasonable rate ~100-500Hz IMU with low noise.
>
> The problem with "bad timestamps" is that they are [...]

Thanks for the reply !

Based on your reply, my understanding is that "a good imu driver output should have stable time intervals.If the actual output imu time intervals appear to be unstable, there may be a problem with the driver or the sensor itself (although based on the clock mark a time, it may not coincide with the actual time it happens), which is not good for any VIO method"

Is that a correct understanding?
Thanks

goldbattle, 2024-01-16:
Correct!

closed 2024-01-16 by goldbattle