ethz-asl/kalibr #659: (Checkerboard corners:1924109): GLib-GObject-WARNING **: 17:13:34.289: cannot register existing type 'GdkDisplayManager'

ZhiYiXinNan, 2023-12-09:
When I use "rosrun kalibr kalibr_calibrate_cameras --target checkerboard.yaml --bag camd435i.bag --models pinhole-equi --topics /color --show-extraction",it shows (Checkerboard corners:1924109): GLib-GObject-WARNING **: 17:13:34.289: cannot register existing type 'GdkDisplayManager'
[image]
what can I do?

goldbattle, 2023-12-31:
Are you running this in a docker or some container that doesn't have a display? You can try to disable showing the final pdf report and it should still save to file ok. Do you have any more details on the OS you running on and what libraries you have installed?

closed 2024-08-12 by goldbattle

506jun, 2025-03-11:
> 当我使用“rosrun kalibr kalibr_calibrate_cameras --target checkerboard.yaml --bag camd435i.bag --models pinhole-equi --topics /color --show-extraction”时，它显示（Checkerboard corners：1924109）：GLib-GObject-WARNING **: 17:13:34.289：无法注册现有类型“GdkDisplayManager” 我该怎么办？ [image]

How was it resolved, please?