ethz-asl/kalibr #662: Please add coordinate system information and clariify meaning of transformations.

lowellm91, 2024-01-03:
I have problem getting your calibration information to work with another piece of SLAM software and the issue I think is related to inconsistent coordinates. In particular in your camera-centric system the x-axis isn't forward, it's to the side which is different than the coordinate system TF2 uses http://wiki.ros.org/tf2/Terminology.

In addition I wanted to clarify what T_cam_imu  does. From what I understand it takes a vector written in the IMU coordinate system to a vector written in cam0 coordinate system, is this correct? So in order to get the location and orientation of the IMU wrt to cam0, I need to take its inverse?

goldbattle, 2024-01-04:
You should expect the camera sensor frame to have the z-axis pointing forward, y-axis down, and x-axis to the right when viewing the from "from behind". This is the convention for computer vision / pinhole camera models.

[image]

Do you see something different? The result pdf should visualize this for you.

lowellm91, 2024-01-09:
Hello, I was not provided the output only the text values used by Kaliber. What about the IMU? In general it would be helpful to have this information on the website under the YAML output section.

goldbattle, 2024-01-16:
The current wiki has what you mentioned:
> T_cam_imu
> IMU extrinsics: transformation from IMU to camera coordinates (T_c_i)

I missed your second question, yes you would take the inverse (there is a closed equation for this inverse), and this would then allow you to take a point in the camera frame and rotate + translate it into the IMU frame.

lowellm91, 2024-01-17:
Ok thank you.

closed 2024-01-17 by lowellm91