ethz-asl/kalibr #667: bad imu sample rate ?

mahammadirfan, 2024-02-02:
Hi,
I am doing cam-imu calibration using dji based m300 drone imu and its front stereo camera, these topics are subscribed from dji osdk ros to onboard nvidia nx board...where i recorded the bag files for kalibr. the imu publishes at 200 hz and both left-right cameras at 20 hz. i tried may times this calibration but not able to implement it with any of the stereo-inertial slam. I observed my imu sample rate is bad......is there anything i can do to improve the kalibr results ?
[image]

Thanks

mahammadirfan, 2024-02-02:
[200hz-imu-results-imucam.txt](https://github.com/ethz-asl/kalibr/files/14143881/200hz-imu-results-imucam.txt)

goldbattle, 2024-03-25:
Your IMU is not something that is usable. You need to fix the driver to address the problem otherwise there is no hope to do calibration or even use the IMU in any down stream application.

closed 2024-08-12 by goldbattle