ethz-asl/kalibr #684: Docker file to install Kalibr in Ubuntu 22.04 couldn't be found in main repo.

AjayAdit, 2024-05-03:
I am trying to install Kalibr in Ubuntu 22.04 with Ros 2 Humble. Could not build the docker files as the Kalibr main repo does not have a docker file for Ubuntu 22.04. Any help on how to configure a docker file or location of docker file would be very helpfull. 

[image]

goldbattle, 2024-05-03:
The build system is only for ROS1. I am not sure if there is any ROS1 on ubuntu 22.04.

AjayAdit, 2024-05-03:
Thanks for the prompt reply, but in Kalibr repo there is a ros 2 documentation. Link https://github.com/ethz-asl/kalibr/wiki/ROS2-Calibration-Using-Kalibr, but its missing a docker file to install it.

Unknown9190, 2024-06-09:
@AjayAdit the meaning of the file docker file name is to choose which Kalibr version that you want to use. For example you can choose to install the ROS Noetic version with Kalibr for Ubuntu 20.04 in your docker images, regardless of what version of Ubuntu you're using.

closed 2024-06-10 by AjayAdit