ethz-asl/kalibr #687: Some issue about reprojection error and imu error

Richardcvpr, 2024-06-04:
The calibtation results is not good enough. I think there may be some issues with IMU, but I don't know where the problem is.

Calibration results
===================
Normalized Residuals
----------------------------
Reprojection error (cam0): mean 4.389178270980999, median 3.4851612083506827, std: 3.546836339123121
Gyroscope error (imu0):
mean 4.987582349959954, median 2.840098285394952, std: 5.739188841909919
Accelerometer error (imu0): mean 5.648619958147809, median 4.5686257632248255, std: 3.9259018299971866
Residuals
----------------------------
Reprojection error (cam0) [px]: mean 4.389178270980999, median 3.4851612083506827, std: 3.546836339123121
Gyroscope error (imu0) [rad/s]: mean 0.04001092025534412, median 0.02278357288981692, std: 0.046040388102236
Accelerometer error (imu0) [m/s^2]: mean 0.7112548280396047, median 0.5752656676632362, std:
0.49433607707441224
Transformation (cam0):
-----------------------
T_ci: (imu0 to cam0):
[[ 0.92757641 -0.27752758 0.25016086 -0.05011873]
[ 0.30536953 0.94890312 -0.07957594 0.01329463]
[-0.2152939 0.15020427 0.96492861 0.33276062]
[ 0.0.0.1.]]
T_ic: (cam0 to imu0):
[[ 0.92757641 0.30536953 -0.2152939 0.11407051]
[-0.27752758 0.94890312 0.15020427 -0.07650671]
[ 0.25016086 -0.07957594 0.96492861 -0.30749456]
[ 0.0.0.1.]]
timeshift cam0 to imu0: [s] (t_imu = t_cam + shift)
-7.978506571910605
Gravity vector in target coords: [m/s^2]
[-0.1185964 0.31407217 9.80080183]Calibration configuration
=========================
cam0
-----
Camera model: pinhole
Focal length: [287.5931203196147, 318.44992107782025]
Principal point: [381.33348613478404, 245.2892815651306]
Distortion model: radtan
Distortion coefficients: [0.014059, -0.021049, 0.001291, 0.003152]
Type: aprilgrid
Tags:
Rows: 6
Cols: 6
Size: 0.021 [m]
Spacing 0.005999999999994 [m]
IMU configuration
=================
IMU0:
----------------------------
Model: calibrated
Update rate: 100.0
Accelerometer:
Noise density: 0.01259165660479
Noise density (discrete): 0.1259165660479
Random walk: 0.000426938443672722
Gyroscope:
Noise density: 0.000802210719501511
Noise density (discrete): 0.00802210719501511
Random walk: 2.82325427421011e-05T_ib (imu0 to imu0)
[[1. 0. 0. 0.]
[0. 1. 0. 0.]
[0. 0. 1. 0.]
[0. 0. 0. 1.]]
time offset with respect to IMU0: 0.0 [s]

[imu_image_good_2024-06-03-17-58-28-report-imucam.pdf](https://github.com/user-attachments/files/15541993/imu_image_good_2024-06-03-17-58-28-report-imucam.pdf)

Can anyone explain me, what's wrong with calibration results?
I would be thankful for any answer.
Thanks

Richardcvpr, 2024-06-04:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

victorsillerico, 2024-06-18:
Some pointers:
1) You have high re-projection error, maybe the intrinsic and extrinsic parameter values are wrong. Did you use Kalibr (from ETH) for camera (only) calibration?
2) The plots of your bias clearly show that the safe bounds 3-sigma were exceed.