ethz-asl/kalibr #689: An error about running a camera calibration

newboylkp, 2024-06-07:
hi，Thank you for reading this question of mine，I think I've successfully installed Kalibr but I got the following error when I run it, I've solved it for 2 or 3 days and still haven't solved it，If you have encountered a similar problem and solved it successfully, please help me, thanks!
[image]
[image]

Andreachen0707, 2024-06-11:
checking by the code:
```
if not isinstance(targetRows,int) or targetRows < 3:
                errList.append("invalid targetRows (int>=3)")
            if not isinstance(targetCols,int) or targetCols < 3:
                errList.append("invalid targetCols (int>=3)")
```
i think you should first check your astra_april.yaml to see what your targetRows and targetCols looks like. If you indeed need targetRow/targetCols less than 3, just modify the code in ConfigReader.py to add a var named errList. It should be fine. 

This does look like a bug. But not a huge one.

newboylkp, 2024-06-11:
> astra_april.yaml
Thanks for your answer! I've found a workaround. I've replaced all the errList in my code with print, so my code runs successfully.
        if targetType == 'checkerboard':
            try:
                targetRows = self.data["targetRows"]
                targetCols = self.data["targetCols"]
                rowSpacingMeters = self.data["rowSpacingMeters"]
                colSpacingMeters = self.data["colSpacingMeters"]
            except KeyError as e:
                self.raiseError("Calibration target configuration in {0} is missing the field: {1}".format(self.yamlFile, str(e)) )
            
            if not isinstance(targetRows,int) or targetRows < 3:
                #print >>targetRows, " ==targetRows "
                print("invalid targetRows (int>=3)")
            if not isinstance(targetCols,int) or targetCols < 3:
                print("invalid targetCols (int>=3)")
            if not isinstance(rowSpacingMeters,float) or rowSpacingMeters <= 0.0:
                print("invalid rowSpacingMeters (float)")
            if not isinstance(colSpacingMeters,float) or colSpacingMeters <= 0.0:
                print("invalid colSpacingMeters (float)")

WFram, 2024-06-11:
Just use AprilGrid board with enough tags (more than 3) and make sure the `.yaml`  file, which describes the pattern and is passed through `--target` when running calibration, contains proper parameters.

goldbattle, 2025-03-16:
Please reopen if this is some issue with Kalibr with more details and how to reproduce.

closed 2025-03-16 by goldbattle