ethz-asl/kalibr #691: My calibrated translation vector is not quite right

SIRIUSwithXU, 2024-06-11:
I had a problem calibrating the camera and imu. The shift vector is too large, and the imu is mounted about 5cm away from the camera. The results are as follows. Does anyone know what affected my calibration results?
[image]
[image]
[image]
[image]
[image]
[MY_NO011-report-imucam.pdf](https://github.com/user-attachments/files/15782967/MY_NO011-report-imucam.pdf)

WFram, 2024-06-11:
Based on angular velocities error plot, IMU model doesn't cover the sensor noise; Same about acceleration bias. I would try to increase the values of these noises and try again. Also, the reprojection errors are large enough. Usually 0.3-0.5 px is fine -> Make sure the camera intrinsics are calibrated properly with an appropriate camera model.

SIRIUSwithXU, 2024-06-25:
> Based on angular velocities error plot, IMU model doesn't cover the sensor noise; Same about acceleration bias. I would try to increase the values of these noises and try again. Also, the [...]

Thank you for your answer. I amplified the noise level and it did get better. And then, mainly because I made a mistake, my accelerometer output is in g, which multiplied by 9.8 is m/2.

closed 2024-08-12 by goldbattle