ethz-asl/kalibr #696: camera extrinsic and re projection matrix

khuechuong, 2024-06-27:
For camera only calibration, does it give extrinsic matrix since I need the re-projection matrix for proper image like in the cameraInfo ros message. The wiki page said it gives extrinsic but it only gives intrinsic. I need to project it so that the black stuff that appears when kalibr undistort it to not appear.

goldbattle, 2024-08-12:
If you only have a single camera then the extrinsic are identity. If you have two cameras, then it will have the SE(3) 4x4 matrix transform between the two cameras in addition to both camera's intrinsics.

khuechuong, 2024-08-12:
@goldbattle so then what would the projection matrix in camerainfo ros message? If I put the same intrinsic in, there are dark everywhere. Is there a node that trims the black stuff around?