ethz-asl/kalibr #699: Visual-inertial calibration on a ground robot

sachaMorin, 2024-07-16:
Hello,

I'm trying to get the transform between a D435 and the IMU on a mobile robot (Clearpath Jackal). I'm following Patrick Geneva's [tutorial](https://www.youtube.com/watch?v=BtzmsuJemgI). I do manage to lift the robot and "excite" all axes of the IMU with rotations. I've tried various bags with small and big motions, sometimes walking around with the robot.

I attached the pdf output of one calibration. The translation is usually close to 0, while I expect something around .4m to .5m on the z axis (in IMU frame).

I was just wondering if anyone had tips for visual-intertial calibration on a ground robot. As this been done successfully before using Kalibr?

On a related note, I think my IMU sampling rate plots are not ideal so this might be the issue. Would this explain a near 0 translation? 

Thanks!

[data-report-imucam.pdf](https://github.com/user-attachments/files/16249341/data-report-imucam.pdf)

goldbattle, 2024-08-12:
How did you get those IMU noise intrinsics. They are very very large. You are sure your IMU is in m/s^2 and rad/sec? Additionally a sample rate <100Hz is likely to fail with the current Kalibr, I am not sure if there will be good performance there.

```
IMU0:
---------------------------Model: calibrated
Update rate: 74
Accelerometer:
Noise density: 0.02129438377
Noise density (discrete): 0.18318121555077344
Random walk: 0.00344152018210807
Gyroscope:
Noise density: 0.00147282089
Noise density (discrete): 0.012669684355875208
````

sachaMorin, 2024-08-19:
I collected still IMU data overnight and used allan_variance_ros then inflated the noise values following the tutorial.

I'll close this since we ended up using another robot with known extrinsics. Thanks for getting back to me!

closed 2024-08-19 by sachaMorin