ethz-asl/kalibr #707: Help：who can briefly talk about the idea of ​​aprilgrid detection and the diff by compared with aprilgtag(https://github.com/AprilRobotics/apriltag)

XavierQL, 2024-08-28:
As I my test shown, the apriltag unit can be detected well by the code from https://github.com/AprilRobotics/apriltag. but it didn't detect any apriltag in aprilgrid by kalibr creat. Meanwile, the kalibr can well detect the aprilgrid.
ps. it didn't cause by the markBorderBits and the image is low distorted (HFOV=30°)
as discripted in paper(AprilTag 2: Efficient and robust fiducial detection), as long as the the apriltag in aprilgrid is segmented, the apriltag in aprilgrid will be detected, am i right? so the border is interrupted by the black squares in the aprilgrid？
I am comfused what's the root cause leading to this big difference? 
Much appriecated

schneith, 2024-08-28:
It has been a long-time since working on it, but iirc we made the border wider than the data bits to reduce misdetections between border and data bits under high distortions / very skewed views.

XavierQL, 2024-08-28:
> It has been a long-time since working on it, but iirc we made the border wider than the data bits to reduce misdetections between border and data bits under high distortions / very skewed views. Aswell as introduced the a checker-board pattern in the AprilGrid to achieve more stable corner detections.

Thx for your reply.
firstly, would you talk about iirc in detail? the border bits of aprilgrid created by kalibr is already 2, this means 2bits border is not enough?
Back to the topic, even i detect the orignal aprilgird image in simulation, apriltag's code doesn't work anyway. there is no distortion, no noise, no view angle, etc.  But if i remove the small black squares, it works well. so i guess the reason is not the border bits, the small black squares did. am i right? 
looking forward to your explaination in detail what the kalibr do to fix this big gap with apriltag's code . much appriecated.

zwiebelbox, 2025-03-20:
Is there a publication available where I could get more information on the design considerations you used when creating AprilGrid? For example about the border width? And I read somewhere that the checkerboard pattern helps with subpixel accuracy, but I couldn't find data for comparison. It seems like it sprung out of nowhere.