ethz-asl/kalibr #722: Interpreting results from IMU-CAM calibration

TL-4319, 2024-11-26:
Thank you for such a great tool. I am performing calibration for my system which includes synchronized cameras and 2 IMUs. I have performed the multi-cam calibration and am now attempting the IMU-CAM calibration which yield these following results and I would like to get some idea on where to begin diagnosing them. Attached is the screenshot of the reprojection error which is not typical from what I have seen from online tutorial and would like to learn what might have caused it. 

[image]

I am also including a link to all the bag file, yam, and calibration results for kalibr below 

https://alabama.box.com/s/732mmgozsjnyzvw72g9qwaz1gdwvbmpq

Thank you in advance

goldbattle, 2025-03-16:
The reprojections should be gaussian (which what you have shown it is not). Likely some motion or IMU issue which is causing this. If you just run camera calibration does it still look like this in the reprojection?

TL-4319, 2025-03-17:
Sorry for not updating but the cause of the issue was due to a driver bug with one of the IMU which flips its axis. No issue with Kalibr.

closed 2025-03-17 by TL-4319