ethz-asl/kalibr #723: Issue with Calibrating Three Overlapping Fisheye Cameras Using Kalibr

fervh, 2024-12-14:
I am having trouble calibrating three fisheye cameras with overlapping horizontal fields of view. The setup includes:
   - One camera at 0 degrees.
   - Two cameras at ±105 degrees.
   - Each camera has a 187° field of view.
   
I have recorded a bag file using an AprilTag calibration pattern of size A0. However, I am stuck while running the rosrun kalibr kalibr_calibrate_cameras command. I’m unsure if my process is correct or if there are specific adjustments or parameters I need to account for when working with fisheye cameras and the A0-sized pattern.

Any help or guidance on properly calibrating these cameras would be greatly appreciated!

LeiyuanMa, 2024-12-16:
are you stuck with this error too?

"[FATAL] [1734339371.433893]: No corners could be extracted for camera /cam0/image_raw! Check the calibration target configuration and dataset."

in the meanwhile, perspective image with no distortion didn't meet this error

goldbattle, 2025-03-16:
Enable showing the detections, and see if the boards are properly detected.
Likely at the edge of the image you might run into detection issues.