ethz-asl/kalibr #726: Problem with plot and generate report -->  Cannot open display

HaenselLeonard, 2025-01-09:
I'm having problem with opening any GUI application in a running container. In short, I always get similar "cannot open display" errors from different applications.

**Platform**
Host : Ubuntu 22.04
Docker : Docker Desktop 4.36.0

Image : kalibr (Ubuntu 20.04 and ROS Noetic based image) 

`who` --> give us  :1
`export DISPLAY=:1`
Run Command
```
xhost +local:root
docker run -it -e "DISPLAY" -e "QT_X11_NO_MITSHM=1" \
    -v "/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    -v "$FOLDER:/data" --net=host      kalibr
```

**I've tried also:**

`xhost +local:root, xhost +local:docker, even xhost +` before running container
`--privileged`
`--runtime=runc`
`DISPLAY=127.0.1.1:1 `(i.e. my host IP)
`DISPLAY=unix$DISPLAY`

**Extra Info**
Some env vars in the container: HOSTNAME=docker-desktop DISPLAY=:1 QT_X11_NO_MITSHM=1

**Inside the docker container:**
```
source devel/setup.bash
rosrun kalibr kalibr_calibrate_cameras \
    --bag /data/cam_april.bag --target /data/april_6x6.yaml \
    --models pinhole-radtan pinhole-radtan \
    --topics /cam0/image_raw /cam1/image_raw
```

Everytime i ran into this error:
```
Generating result report...
Traceback (most recent call last):
  File "/catkin_ws/devel/lib/kalibr/kalibr_calibrate_cameras", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 465, in <module>
    main()
  File "/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_calibrate_cameras", line 429, in main
    kcc.generateReport(calibrator, reportFile, showOnScreen=not parsed.dontShowReport, graph=G, removedOutlierCorners=removedOutlierCorners);
  File "/catkin_ws/src/kalibr/aslam_offline_calibration/kalibr/python/kalibr_camera_calibration/CameraUtils.py", line 428, in generateReport
    fig = pl.figure(offset)
  File "/usr/lib/python3/dist-packages/matplotlib/pyplot.py", line 539, in figure
    figManager = new_figure_manager(num, figsize=figsize,
  File "/usr/lib/python3/dist-packages/matplotlib/backend_bases.py", line 3259, in new_figure_manager
    return cls.new_figure_manager_given_figure(num, fig)
  File "/usr/lib/python3/dist-packages/matplotlib/backends/_backend_tk.py", line 950, in new_figure_manager_given_figure
    window = tk.Tk(className="matplotlib")
  File "/usr/lib/python3.8/tkinter/__init__.py", line 2270, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":1"
```

In addition, even if I set `-dont-show-report` as a parameter, I always get the same error. I have also tried 'xeyes' as a test. Here I also get the same error regarding `Can't open display: :1`. I am beginning to despair. Do you have a solution?

anton-pintya, 2025-02-24:
Same problem, but for [stereolabs/kalibr](https://hub.docker.com/r/stereolabs/kalibr) image (host machine: Ubuntu 22.04). According to Usage from the link I need to run `xhost +local:root` before running container, but still while running imu_camera calibration I recieve error:
```shell
File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1818, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"
```
So I get final `*.yaml` and `*.txt` files, but without report with plots it's difficult to estimate "accuracy" of this calibration. Did you succed in solving this problem?

goldbattle, 2025-03-16:
There should be no issue with running with `--dont-show-report`. Just make sure you do not have the verbose, or the visualize detections enabled also.

goldbattle, 2025-11-16:
You can try specify a non-interactive backend for matplotlib via this before your command.
```
export MPLBACKEND=Agg
```