ethz-asl/kalibr #738: I want to obtain the pose of each image.

jxhoh, 2025-03-12:
As a result, I only obtained the final internal parameters of the camera. I want to obtain the external parameters of each image taken relative to the tag. What should I do?

goldbattle, 2025-03-16:
I believe the `--export-poses` option is what you are interested in?

jxhoh, 2025-03-17:
> I believe the `--export-poses` option is what you are interested in?

Yes, this is useful. However, I want to use this pose for the hand-eye calibration algorithm. Is the timestamp of this pose corresponding to the bag file generated by my SLAM system? Or, how can I obtain the pixel points and target points that it has recognized?

jxhoh, 2025-03-17:
> I believe the `--export-poses` option is what you are interested in?我相信 `--export-poses` 选项是您感兴趣的？

I used the exported pose to perform re-projection calculations, but I couldn't obtain such a low re-projection error as that in Karlib. I'm not sure if the pose is not aligned with the time in my bag file. I don't know the corresponding relationship between the time of this pose and the time in the bag file I provided.